mikedmor/OctoPrint_MultiCam #1: Octoprint1.3.8 : Webcam don't appear

jjtronics, 2018-05-10:
Hello,
Great idea this plugin!
I already have two webcams running on my octoprint 1.3.8 and I installed the latest version of MultiCam 0.2.1 But i have no window of visualization and only one webcam appears in control tab : 
https://www.dropbox.com/s/7ivp45ovb321qsv/Capture%20d%27%C3%A9cran%202018-05-10%2004.36.25.png?dl=0

I have two webcam in multicam parameters : https://www.dropbox.com/s/mc3cczddxt082yq/Capture%20d%27%C3%A9cran%202018-05-10%2004.35.34.png?dl=0

and in my .octoprint/config.yaml , i have : 
  multicam:
    multicam_profiles:
    - URL: /webcam1/?action=stream
      name: Default
    - URL: http://192.168.1.44:8081/?action=stream
      isButtonEnabled: true
      name: Webcam 2

Another person has the same problem?

Thank you in advance,

mikedmor, 2018-05-10:
Looks like for some reason your first webcam did not get the isButtonEnabled settings. If you add that onto the Default webcam everything should start working again for you. 

`multicam:
multicam_profiles:
- URL: /webcam1/?action=stream
isButtonEnabled: true
name: Default
- URL: http://192.168.1.44:8081/?action=stream
isButtonEnabled: true
name: Webcam 2`

0.2.1 was suppose to fix the data migration issue, but it still seems to be a problem, ill look into it some more and try to get another update to fix it. Please let me know if that solves your issues.

mikedmor, 2018-05-10:
I just pushed 0.2.2. This should have a fix for the data migration issues. However it may reset you settings back to the new defaults. You may be able to avoid the reset if you add _config_version: 1 like the snipplet below before updating:

```
multicam:
    _config_version: 1
    multicam_profiles:
    - URL: /webcam1/?action=stream
       isButtonEnabled: true
       name: Default
    - URL: http://192.168.1.44:8081/?action=stream
       isButtonEnabled: true
       name: Webcam 2
```

jjtronics, 2018-05-10:
Hello,

Thank you for your help !
Now the camera menu is displayed correctly but I still do not have a video window : 
https://www.dropbox.com/s/ff9b0cia9zbwequ/Capture%20d%27%C3%A9cran%202018-05-10%2017.38.19.png?dl=0
I update in 0.2.2 but the result is the same : 
https://www.dropbox.com/s/vj7yfuhaygdhamp/Capture%20d%27%C3%A9cran%202018-05-10%2017.40.46.png?dl=0

Here are the parameters MultiCam profiles : 
https://www.dropbox.com/s/9hl425ly57o3y1z/Capture%20d%27%C3%A9cran%202018-05-10%2017.41.16.png?dl=0
I test with the IP and the local path

Thank you in advance ;)

mikedmor, 2018-05-10:
Hmm... What plugins do you have installed? I know there is one that moves the webcam to a tab of its own, that or something else may be interfering with my plugin.

jjtronics, 2018-05-10:
Yes, i use webcam tab.
I just disable it and it works now !!
Thank you !!

mikedmor, 2018-05-10:
No problem! Glad everything is working now! Ill add a notice about webcam tab to my repo.

closed 2018-05-10 by mikedmor