mikedmor/OctoPrint_MultiCam #2: Camera rotation

Darkmadda, 2018-05-12:
We need an option for rotating the view like how the primary camera can be rotated in the main settings

mikedmor, 2018-05-12:
I will look into adding a rotation setting in one of the next few updates. I'm currently running from an external pi that handles all of that for me before the feed is streamed. But i can see the need for it if you are using a camera directly on the pi.

jeremyrclark, 2019-12-20:
I'd like to second this request. I use OctoPi with a secondary RaspiCam that is mounted upside down. I use OctoPod to view my feeds remotely and it isn't able to rotate or flip the camera via settings as it relies on whatever MultiCam provides.

mikedmor, 2020-06-17:
Version 0.2.8 should now include options for rotating and flipping the camera.

closed 2020-06-17 by mikedmor