mikedmor/OctoPrint_MultiCam #3: Camera Buttons do not always re-enable when clicking between veiws

PixlEmly, 2018-05-17:
If you switch between cameras a few times, on occasion both buttons will become stuck disabled, requiring a page reload to unstick them - simply switching tabs will not restore functionality.

mikedmor, 2018-05-18:
Could you double check which version you are running in the Plugin Manager. If its not 0.2.6 try updating, otherwise try to refresh the page and see if toggling webcams triggers this. If not open up the settings panel and it save, did that break it? 

0.2.6 should not have the saving bug any longer. But if it is happening for you if you could also press F12 and send any errors in the console tab. As well as your browser version. Then i can debug this better.

mikedmor, 2023-06-05:
Fixed in version 0.4.0rc3