mikedmor/OctoPrint_MultiCam #4: Show Multiple Cams - Feature Request

gernei09, 2018-07-29:
Any progress on the features mentioned below:

“Future updates will include the more options to show different types of streams, as well as the abilitly to show more than one stream at a time.”

I’m particularly interested in the showing multiple streams part, which I’d find very useful.

If you have any unfinished code related to it, I wouldn’t mind assisting with completing it.

mikedmor, 2018-08-01:
I have not actually started working on that yet, but i probably wont be able to look into it for another month or so since i'm in the middle of moving at the moment. I marked this as an enhancement request and will get back to it asap. 

Otherwise feel free to pull the code and submit the change back to me, when i find some time i can review it and merge it back into the repository if everything looks good.

mikedmor, 2018-08-01:
Also wanted to note that if you decide to submit the change back to me ill be sure to give you credit for the update and link your github (if you are interested) on the readme here and for the repository listing. 

Thanks for the offer! Extra help is always welcomed, and any feedback too. Sometimes its hard to see the full potential of your project without a little extra help!

n3roGit, 2020-11-24:
Yes Plase add this feature!

wolfgangr, 2024-03-09:
I' like to add weight to the pledge, and even suggest a browser view with multiple (in my case 2) cams and the GCode viewer in one frameset.

I have my cams setup to 320x240 pixel anyway, to keep CPU load low (run on single core 1.2 GHz x86). I think details are less important than machine overview.

Thanks for your great work!