mikedmor/OctoPrint_MultiCam #5: Selected cam reverts to default upon switching tabs

delamitry, 2018-08-01:
The selected camera reverts to the default camera when switching between browser tabs without acknowledging the change. (i.e. button still shows previously selected cam depressed)
I would think the expected behavior would be to persist the selected camera when switching between browser tabs. 
v0.2.6

mikedmor, 2018-08-01:
Some recent updates might have broken that.

By default the camera always returns to the original cam when switching tabs, but the buttons should also reflect that. When i get some time ill revisit that and see if i cant get it working again. Thanks for the bug report.

Viginox, 2018-09-05:
I encountered the same issue on the web interface as well as on TouchUI on the RPI3B+.
But an interesting situation occurs, when only the second camera is connected to the Raspi. Then you can not change to this camera in the Web Interface, but when using TouchUI you can switch to it for a very short time. After that, it resets to the default camera.

ndupont-net, 2018-12-26:
Hi, I confirm the same behavior, which is not too bad in the full web interface but renders it useless in TouchUI.

totalitarian, 2019-07-25:
Same issue here too :(

timur-tabi, 2020-03-08:
> I encountered the same issue on the web interface as well as on TouchUI on the RPI3B+.
> But an interesting situation occurs, when only the second camera is connected to the Raspi. Then you can not [...]

I believe this is a TouchUI bug.  TouchUI doesn't support MultiCam.  See https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/371

BillyBlaze, 2020-03-09:
The OP's issue doesn't seem like a TouchUI bug to me but default OctoPrint behaivor:

1. Changing the [browser tabs](https://github.com/foosel/OctoPrint/blob/6232a04e52fbf09a8b912590718764fccc3598ba/src/octoprint/static/js/app/viewmodels/control.js#L434) or [OctoPrint tabs](https://github.com/foosel/OctoPrint/blob/6232a04e52fbf09a8b912590718764fccc3598ba/src/octoprint/static/js/app/viewmodels/control.js#L426) will invoke an event called 'onTabChange'.
1. That 'onTabChange' calls [disableWebCam](https://github.com/foosel/OctoPrint/blob/6232a04e52fbf09a8b912590718764fccc3598ba/src/octoprint/static/js/app/viewmodels/control.js#L364) when going away from OctoPrint or Controls Tab and reset the WebCam image to an empty stream
1. When going back to OctoPrint or Controls Tab the 'onTabChange' calls [enableWebCam](https://github.com/foosel/OctoPrint/blob/6232a04e52fbf09a8b912590718764fccc3598ba/src/octoprint/static/js/app/viewmodels/control.js#L381) and restores it back to the OctoPrints default webcam image.

Like @mikedmor said, the fix to the OP's is either updating the button back to the correct state. Or create a hooks that will set the webcam stream directly to previously selected webcam.

@timur-tabi, is this what you mean with support for MultiCam for TouchUI?

ttabi, 2020-03-09:
My point is that TouchUI does not appear to recognize that the webcam buttons are associated with the camera view.  TouchUI separates the camera into its own tab, distinct from the Control panel tab.

BillyBlaze, 2020-03-09:
Yeah, that is what I assumed the feature request was about, but since it was linked to another bug I needed clarification that we are on the same page.

shaelr, 2020-03-13:
> e fix to the OP's issue is either upda

would you happen to know how create said hook?

BillyBlaze, 2020-03-15:
That is something for inside the plugin. I would just override the value of settings.webcam_streamUrl when changing a webcam stream instead of overriding the image source directly.

shaelr, 2020-03-15:
@mikedmor any updates with this? would be nice to have this feature working?

AbuMaia, 2020-06-16:
I'd like to second the request for updates. This is quite the annoying bug.

mikedmor, 2023-06-05:
Fixed in version 0.4.0rc3