mikedmor/OctoPrint_MultiCam #9: remote cams not available remotely

bryanhunwardsen, 2018-09-11:
I have motioneyeos on a second pi serving several streams to octopi via this excellent plugin, however, they are only viewable on the local network and not remotely(as the primary stream from the octoprint is). Im not sure if this is a setup issue with the motioneye cams address, something required in haproxy not listed with this plugin or some other issue. (So it may not really be a bug)

My remote cams are setup in multicam i.e. 192.168.1.125:8081
It seems the plugin is just forwarding the address and not essentiallially repackaging it for external consumtion via octoprint remote sessions on port 80 etc, if this is a bug, please consider a fix, if it is my config, please suggest remediation, if it is a feature enhancement, please consider.

Thx

redxeth, 2018-09-20:
I see this as well.  Not sure how to access if using second port.   Will have to figure out a non-port type solution perhaps so I can leave off the local IP #.

bryanhunwardsen, 2018-09-20:
I filed similar on octoprint and Gena responded that remediation would have to happen at haproxy level, so maybe additional haproxy config instructions, or that plus some tweek on the plug-in to support haproxy config.

redxeth, 2018-09-20:
I got it working last night with haproxy hack.  Posted it on the Facebook Octoprint Users Group.

Here is the rub: 

[image]

ndupont-net, 2018-12-26:
Hi, it also works for me with the same haproxy settings.

But make sure you don't enter any IP-address in the Multicam config.

I.E.  :   /webcam2/?action=stream   and not http://127.0.0.1/webcam2/?action=stream or any http://192.168.0.10/webcam2/?action=stream seen on the excellent Chris Basement tutorial on YouTube.

harryurban, 2019-05-02:
That’s not a haproxy ‘hack’, it’s properly added configuration. 😉

closed 2020-06-17 by mikedmor

hobramycin, 2020-06-17:
I corrected this issue for myself by restreaming the existing stream on the pi running octopi using the following script:

```
#!/usr/bin/env bash

# you can either make a separate path or port for each stream
# depending on your choice, your streams will be available at
# http://localhost/$BASE_WWW_PATH/?action=stream or
# http://localhost:$OUTPUT_PORT/?action=stream

MJPGSTREAMER_HOME="${HOME}/mjpg-streamer/mjpg-streamer-experimental"
BASE_WWW_PATH="www"
WWW_PATH="${MJPGSTREAMER_HOME}/${BASE_WWW_PATH}/"
INPUT_PLUGIN="input_http.so"
INPUT_HOST="motioneye.local" # motioneye server address
INPUT_PORT="8081" 
INPUT_STRING="${INPUT_PLUGIN} -H ${INPUT_HOST} -p ${INPUT_PORT}"
OUTPUT_PLUGIN="output_http.so"
OUTPUT_PORT="8083" 
OUTPUT_OPTS="-w ${WWW_PATH}"
OUTPUT_STRING="${OUTPUT_PLUGIN} -p ${OUTPUT_PORT} ${OUTPUT_OPTS}"

stopStream() {
    local pids
    pids=$(jobs -pr)
    if [ -n "$pids" ]; then
        kill "$pids" || killall -s2 -e mjpg_streamer
    else
        killall -s2 -e mjpg_streamer
    fi
    logger -s "$0: Successfully killed mjpg_streamer process, exiting" && exit 0
}

startStream() {
    pushd "${MJPGSTREAMER_HOME}" > /dev/null 2>&1 || { logger -s "Failure to pushd to mjpg directory"; exit 1; }
        logger -s "Running \"$( which mjpg_streamer )\" -i \"${INPUT_STRING}\" -o \"${OUTPUT_STRING}\""
        LD_LIBRARY_PATH=. /usr/local/bin/mjpg_streamer -i "${INPUT_STRING}" -o "${OUTPUT_STRING}" &
        sleep 1 &
        sleep_pid=$!
        wait ${sleep_pid}
    popd > /dev/null 2>&1 || { logger -s "Failure to popd out of mjpg directory"; exit 1; }
}

startStream
trap "stopStream" SIGINT SIGTERM
while true; do
    sleep 1
done
```