mikedmor/OctoPrint_MultiCam #12: How to find second camera adress?

CaptainKiwii, 2019-03-03:
Hi
Ok so kind of new in here and with plugins, I could get how to find my second camera (connected with usb on the rpi) adress.

I have a c920 pro connected and set as default, but the c170 isn't showing.

So how am I supposed to make this work? I found in one of the other thread that commands :

https://user-images.githubusercontent.com/5686085/45827289-b8caba00-bcbb-11e8-85fb-c4d4665b8f18.JPG

Yet I found that the haproxy thing tend to slow things down a lot.

https://github.com/foosel/OctoPrint/issues/1846

So... not sure what I'm supposed to do.

What I want to do is using my C920 for timelapse purposes with octolapse, having it above the printer while I have the c970 attached on the z axis for monitoring purpose (feed wouldn't have to be stored, just displayed for that one).

What am I misssing ?

harryurban, 2019-05-02:
I would say that there is no stream running for the second webcam so you need to start a second instance of m-jpg streamer with the correct device like /dev/video1.

CaptainKiwii, 2019-05-02:
Hi, first of all thanks for taking the time to answer to you, any idea of how to start that second instance please? (even link or direction)

harryurban, 2019-05-03:
well just to get the stream running, this can be achieved like so:
```cd $HOME/mjpg-streamer/
./mjpg_streamer -o "output_http.so -w ./www -p 8081" -i "input_uvc.so -d /dev/video0 -f 10 -r 1280x720"```

see that used video0, thats my usb cam
the raspicam. which is my first cam is handled by the original mjpeg stream.

also i changed the port to `8081`.

if that works you need to automatically find a way to start the stream, with things like init.d scripts ,nohup, rc.local etc.

if  you need to see both stream on one page, you need to add a second backend to the haproxy, so that everything is reachable on port 80.

can be achieved like so:

```
 frontend public
        bind :::80 v4v6
        bind :::443 v4v6 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        use_backend webcams if { path_beg /webcams/ }
        use_backend topcam if { path_beg /topcam/ }
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint

backend topcam
        reqrep ^([^\ :]*)\ /topcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8081
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http

backend webcams
        mode http
        errorfile 503 /etc/haproxy/webcams.html
```

and `cat /etc/haproxy/webcams.html` 

```
<html>
<head>
    <style>
        img {
            display: block;
            width: auto;
            height: auto;
            max-width: 100%;
            max-height: 90%;
        }
    </style>
</head>

<body>
    <div>
        [image]
        [image]
    </div>
</body>

</html>
```

harryurban, 2019-05-03:
http://www.thedoble.com/3d-printing/running-two-webcams-with-octoprint-for-hd-timelapses/

CaptainKiwii, 2019-05-03:
Thanks again ;)
I'm diving into that once I got back from work.