mikedmor/OctoPrint_MultiCam #13: Second Webcam URL will be ignored

tis-cs, 2019-03-11:
Hi,

I'm using a Axis ip camera to see what is happening while printing.
Now I tried to use the pi cam via mjpeg-streamer, which is for my timelapse, to set it as a second cam.

[image]

But all I get is a error which shows that the plugin tries to stream the Axis cam:

[image]

What am I doing wrong or is it a bug?

Thanks in advance!

BR

harryurban, 2019-05-02:
Hast du das Problem schon gelöst?
Siehst du was im Netzwerk tab von den dev Tools? Ein 401 oder ähnliches oder CORS Probleme?

tis-cs, 2019-05-03:
ehrlich gesagt habe ich einfach die Pi-Cam durch eine Logitech USB Cam getauscht. Damit läuft es, allerdings nicht mit der 127.0.0.1 sondern ich muss die IP des Raspi explizit angeben.

harryurban, 2019-05-03:
hast du mal probiert host / ip wegzulassen?

also `/webcam/?action=stream` als url einzugeben?

tis-cs, 2019-05-03:
habe ich eben gemacht. Geht soweit mit der USB Cam. Raspi-Cam hängt aktuell ja nicht mehr dran.
Ist aber auch nicht mehr wirklich interessant, da die Qualität der Cam nicht gerade gut ist.

harryurban, 2019-05-03:
hab hier bisschen meine anleitung für usb webcam + picam aufgeschrieben falls du doch noch beide anschliessen willst:
https://github.com/mikedmor/OctoPrint_MultiCam/issues/12

harryurban, 2019-05-03:
http://www.thedoble.com/3d-printing/running-two-webcams-with-octoprint-for-hd-timelapses/

BramBertels, 2019-05-10:
I had the same issue. Removing the ip address and just using /webcam/?action=stream and /webcam2/?action=stream worked for me

mikedmor, 2020-06-17:
@BramBertels is correct, If using a local webcam plugged into the pi, then do not include the address, instead just use /webcam2/?action=stream

@harryurban also has a great write up on how to get everything configured.

closed 2020-06-17 by mikedmor