mikedmor/OctoPrint_MultiCam #31: config.yaml second camera data overwritten with first camera data

wseifert, 2020-09-17:
in config.yaml sometimes the data for the second cam are overwritten with data from first cam, so both Webcam buttons püoint to the same first cam.

YoungCereal, 2020-09-23:
Same at me all cams are overwritten

ProfEngr, 2020-11-01:
Mine gets stuck frequently on the webcam2 URL in the Octoprint webcam settings. It's easy to correct, but the randomness is odd. It's done it mid-session and after a reboot.

n3roGit, 2020-11-23:
Here the same

n3roGit, 2020-12-09:
As a workaround create the first cam with the addon "double". Then you have one of the buttons double, but nothing will be overwritten.

[image]

closed 2023-06-05 by mikedmor