mikedmor/OctoPrint_MultiCam #32: Switching between cams on Safari does not work

nibty, 2020-11-18:
Only the default cam works on Safari. Click the button for other cams does nothing.

OS: Mac OS 11.0.1 (20B29)
Browser: Safari Version 14.0.1 (16610.2.11.51.8)

HDRI00, 2020-11-24:
I second this problem.

OS: Mac 10.14.6
Browser: Safari Version 14.0.1 (14610.2.11.51.10)

spatcho, 2020-11-29:
Same here, Big Sur x86, doesn't work on Safari, work on Chrome and Firefox

nibty, 2020-11-30:
I found the origin of the issue in the Octoprint code. Removing this block of code fixes the problem.

https://github.com/OctoPrint/OctoPrint/blob/f67c15a9a47794a68be9aed4f2d5a12a87e70179/src/octoprint/static/js/app/viewmodels/control.js#L684-L687
https://community.octoprint.org/t/mjpeg-streaming-in-safari-possible-improvements/16064

```js
// safari bug doesn't release the mjpeg stream, so we just set it up the once
if (OctoPrint.coreui.browser.safari && currentSrc != undefined) {
    return;
}
```

victor7376, 2020-12-10:
Just to inform, this also is a fix if you’re using an iPad.

rshartog, 2021-05-31:
Confirmed. Switching cameras in OctoPrint works for me in Chrome, but did not work in Safari. Commented lines 684-687 as mentioned by nifty above, reloaded OctoPrint, and camera switching works in Safari. Thanks for finding this. I spent hours trying to understand why I could see my cameras with local URLs and full URLs, but the MultiCam buttons on the Control tab would only ever show the first camera (in Safari). I never thought to try another browser.

appleimperio, 2021-07-13:
Im having the same problem, where is the file I need to edit?

Thanks

victor7376, 2021-07-13:
> I'm having the same problem, where is the file I need to edit?
> 
> Thanks

Location:

/home/pi/oprint/lib/python3.7/site-packages/octoprint/static/js/app/viewmodels/

appleimperio, 2021-07-13:
Thank you. I can't find but I think its because im using a docker container!

victor7376, 2021-07-14:
> Thank you. I can't find but I think it's because im using a docker container!

Just look for control.js then look for the line regarding safari.

appleimperio, 2021-07-14:
Found it. just in case someone in Docker came looking for it the path of the control.js file is:

/usr/local/lib/python3.8/site-packages/octoprint/static/js/app/viewmodels/

Thank you again it work great!

levydanqc, 2023-03-05:
Why hasn't this fix been merged ? It is quite annoying as commenting these lines fix the problem...

victor7376, 2023-03-05:
> Why hasn't this fix been merged ? It is quite annoying as commenting these lines fix the problem...

Never understand why they never remove it, I keep having to remember to remove it for my octoprint to work.

mikedmor, 2023-06-05:
Fixed in version 0.4.0rc3

closed 2023-06-05 by mikedmor