mikedmor/OctoPrint_MultiCam #46: multicam not able to reach video - but link on error page works?

gleep52, 2022-02-06:
When I fill in the values to my mjpg stream (no password required) I can paste it in any browser and it pops up the mjpg stream instantly.  I am using the openmiko firmware on my Wyze V2 camera which is supposedly made for octoprint.  

Multicam fails to pull the video and brings up an error page with a link to the video it is trying and that link, when clicked from the error page, works in the tab it opens.  Why can't multicam pull the video feed if it's working?

If I test the snapshot url it works - and returns a static pic.  If I test the stream it never connects and populates the test page.  

https://github.com/openmiko/openmiko  is the firmware I'm using.  

The values I put into multicam are here:
http://192.168.3.215:8080/?action=snapshot
http://192.168.3.215:8080/?action=stream

But it fails to load the stream - any ideas why?

OllisGit, 2022-02-09:
Hi @gleep52,

please check via browser developer tools, if the mentioned url is the same in ``` <img src=```. Maybe there is a "illegal whitespace".
Is the Webcam working the OP-Settings page "Webcam & Timelapse"

[image]
 
Please paste a screenshot of the error-message that appears. Maybe it is an error page from OP and not from the Plugin

My current assumption is, that OPs Test-Function expect a different response from the camera stream (e.g. no content-type):
[image]

gleep52, 2022-02-12:
I do not see any white spaces in the code... here is the page I am referring too.  It does NOT work on the normal webcam page either... openmiko firmware was supposed to be octoprint compatible natively but I seem to have issues with it's fluidity of playback and wifi connection.  I am now using the standard RTSP firmware from Wyze directly and have it in Blue Iris again - that's the URL listed in this screenshot:
https://www.screencast.com/t/74kDNcqXBnSM

That exact link on that page is clickable - and it also streams from VLC with that same info...

OllisGit, 2022-02-14:
Hi @gleep52,

the screencast-image shows a different url like mentioned in the first comment.
If you open the url ```http://192.168.1.5:8999/mjpg/ddd``` in a  browser tab the stream is visible, right? Also with the ```ddd``` at the end (never saw this before) ?

gleep52, 2022-02-15:
This is kind of merged with the other issue I opened.  I’m back to using blue iris to generate my mjpg stream since it’s much more reliable than the openmiko firmware.  With blue iris you tell it which stream type then a slash and the cameras short name… ddd = 3D ( for my 3d printers).  ;)

OllisGit, 2022-02-15:
hmmm...running out of ideas.
Please try to call your cam stream via curl with verbose command. It should look something like this and should at least include the status, content-type
```
curl -v http://192.168.178.37:8081/\?action\=stream

*   Trying 192.168.178.37:8081...
* Connected to 192.168.178.37 (192.168.178.37) port 8081 (#0)
> GET /?action=stream HTTP/1.1
> Host: 192.168.178.37:8081
> User-Agent: curl/7.77.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Access-Control-Allow-Origin: *
< Connection: close
< Server: MJPG-Streamer/0.2
< Cache-Control: no-store, no-cache, must-revalidate, pre-check=0, post-check=0, max-age=0
< Pragma: no-cache
< Expires: Mon, 3 Jan 2000 12:34:56 GMT
< Content-Type: multipart/x-mixed-replace;boundary=boundarydonotcross
< 
--boundarydonotcross
Content-Type: image/jpeg
Content-Length: 21548
X-Timestamp: 113.375621

Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
* Failure writing output to destination
* Closing connection 0

```

Did you take a look into the Browser-Console, maybe there is a CORS-Issue?
Maybe your cam did not response with ```Access-Control-Allow-Origin: *```, then the browser did not allow to communicate with this "external-ressource". 
Your OP-Views are loaded from one "location" and the stream is loaded from a different one.

gleep52, 2022-02-17:
When I curl my stream I get this output:

```
curl -v http://192.168.1.5/mjpg/ddd
*   Trying 192.168.1.5...
* TCP_NODELAY set
* Connected to 192.168.1.5 (192.168.1.5) port 80 (#0)
> GET /mjpg/ddd HTTP/1.1
> Host: 192.168.1.5
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: BlueServer/5.5.5.6
< Date: Thu, 17 Feb 2022 04:53:41 GMT
< P3P: CP="CAO COR CURa ADMa DEVa OUR IND ONL COM DEM PRE"
< Access-Control-Allow-Origin: *
< Set-Cookie: session=6904233311420b3576454468719f2406; path=/
< Connection: close
< Cache-Control: no-cache, no-store
< Content-Type: multipart/x-mixed-replace;boundary===STILLIMAGEBOUNDARY==
< 
Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
* Failed writing body (0 != 2533)
* Closing connection 0

```
I do not believe there is any kind of redirection of video going on.  My curl response has the allow origin from anywhere... 

Do you see anything else in my curl output that could help pin point why it doesn't work?

MartinRusk, 2022-03-04:
I have exactly the same issue here - the browser is giving a CORS policy error. 
`Access to image at 'http://prusacam:8081/' from origin 'http://octopi' has been blocked by CORS policy: The request client is not a secure context and the resource is in more-private address space.`
How could that be adressed in MS Edge?

anthonielamay, 2023-03-03:
Showing a similar (if not the same) issue.  Was there ever a resolution for this?
[image]