mikedmor/OctoPrint_MultiCam #51: Can't edit default camera or change which camera is default

stayingarmed, 2022-12-16:
I'm using an old Galaxy Note 8 to host Octo4a to run octoprint and everything has been working perfectly (It uses the camera phone to stream with) but I recently got a usb webcam that I hooked up for a different view. 

The problem is, Multicam locks the default camera (the camera on the phone) so I can't change it to default to the usb camera instead.....is there a way to change what camera is set as default? 

Only reason I want to change is I'm also using Obico (previously known as Spaghetti Detective) and it doesn't allow selecting from different cameras and only uses the default camera, but I want the default camera to be using for Obico to be the usb camera.

Is this possible?

QuePID, 2022-12-17:
I resolved the issue. Instead of using the info from /dev/v4l/by-id/ I used the info from /dev/v4l/by-path which showed two index-o items.

platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.2:1.0-video-index0
platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.4.2:1.0-video-index0

By using these in my webcam2 and webcam3 txt files I was capable of getting three webcams working (2 usb, 1 raspi).

kevman28, 2023-01-04:
same issue here, I want to be able ot change the default but it is locked in hard coded.

mikedmor, 2023-01-04:
Hey Guys. Just wanted to let you know that this is not a bug. The default cam is configured via octoprints webcam settings. If you go to the webcams and timelapses tab in octoprint you can configure the default from there. The plugin uses its first slot from that page, the only reason it is there is so that you can give it a name for when you want to switch back to it. Good luck!

closed 2023-01-18 by mikedmor