mikedmor/OctoPrint_MultiCam #53: Multicam seems to be broke for octoprint 1.9.0RC2

swheettaos, 2023-03-11:
I know they are making major changes to how the webcam works.  I noticed my webcams aren't working.  I have to run them in 4:3 normally the problem is in your plugin (and the new octoprint web interface) it doesn't seem to give me that option anymore.

RoboMagus, 2023-03-29:
Can confirm the plugin is no longer functional with `1.9.0`.

Web UI Console provides the following:
```
Please use the webcam system introduced with 1.9.0, the webcam_snapshotUrl config setting is deprecated and will be removed in a future release. Stacktrace: Error
    at SettingsViewModel.read (https://octoprint.my-domain.com/static/webassets/packed_core.js?f195a60e:342:137)
    at Function.yd (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:18275)
    at Function.zd (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:17972)
    at Function.ha (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:17808)
    at Function.Hc (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:17653)
    at Function.notifySubscribers (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:9667)
    at Function.xa (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:12024)
    at Object.c [as snapshotUrl] (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:11759)
    at SettingsViewModel.write (https://octoprint.my-domain.com/static/webassets/packed_core.js?f195a60e:342:361)
    at SettingsViewModel.e [as webcam_snapshotUrl] (https://octoprint.my-domain.com/static/webassets/packed_libs.js?6bee1dee:130:15710)

````

crysxd, 2023-04-30:
I was involved in redesigning the webcam stack for 1.9.  The first stack trace is a warning and not cause of the issue, OctoPrint 1.9 comes with a compat issue to prevent old plugins from breaking.

The second stack trace is what breaks the plugin. The plugin seems to meddle with OctoPrint's UI, which changed with 1.9.

@mikedmor Do you plan on supporting OctoPrint 1.9?

mikedmor, 2023-05-01:
Hey All, Its been some time since i last worked on this as its been functional for me up until 1.9. I'm in the middle of a few other projects, but still use this plugin myself daily so I definitely have intentions to update it, but at the moment I do not have a timeline for when that will happen. Ill update this thread again once i start working on the adjustments, and have an update available.

crysxd, 2023-05-02:
@mikedmor I'm happy to assist you if you are still willing to maintain this plugin. I don't have a working dev environment right now as I'm not home until the 11th of May, but can help you after with a PR.

I think the conversion is super simple in the end:
- Add `WebcamPlugin` to your plugin
   - Implement the function to return the camera configs. This is a rather simple mapping of the plugin settings to the well defined structure
   - Implement the "get snapshot" function (plugins are responsible to obtain snapshots). This can be copied 1:1 from the bundled "classic webcam" plugin
   - Define a Jinja template to be shown in the webcam area in OctoPrint's UI. Again, can be copied from "classic webcam". Only thing to figure out here is how to pass the webcam id to the plugin, but most likely this can be done by looking at the html element's id
 - Remove any code that was used before to show your webcams in OctoPrint's UI

M-GBear, 2023-05-24:
Yes Multicam is dead with the release of 1.9

Hope @mikedmor and @crysxd can get together soon for a fix, especially as I've spent many hours configuring my cams to only have them lost to the upgrade a weeks later :(

Really appreciate the effort @mikedmor has put in to maintaining this plugin for a long time.

Thank you.

crysxd, 2023-05-27:
I made a first rough implementation. Doesn't show the webcam yet, but the webcams are now recognized by OctoPrint. We now need to create a surface to show the webcam in (same as the bundled plugin) and apply rotation etc

[image]

[image]

dirkvranckaert, 2023-05-28:
Any progression the issue and maybe a targettes Release date?

M-GBear, 2023-06-01:
How easy is it to downgrade back to the previous version of Octoprint? Will doing that restore the previous functionality with Multicam, or will I need to setup everything again?

mikedmor, 2023-06-01:
Hey @M-GBear You could try downgrading this way: https://community.octoprint.org/t/how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image/205

I think anything before 1.9.0 should still be working. I have been working on the changes to get this setup, but its been going a little slow. Hopefully I can get things wrapped up and post an update over the next few weeks. Good luck downgrading.

M-GBear, 2023-06-01:
@mikedmor thanks for that link, it seems straight-forward enough. I'm in the middle of a 10 hour print job so I'll tackle this tomorrow. I'll be sure to let you know how it goes. Also, great news that you hope to post an update within weeks. Cheers

chrisjq, 2023-06-01:
Here's a patch to get it going from master:
[OctoPrint_MultiCam-56dcdb8-feat: Resolve multicam issue in latest octoprint.patch](https://github.com/mikedmor/OctoPrint_MultiCam/files/11623323/OctoPrint_MultiCam-56dcdb8-feat.Resolve.multicam.issue.in.latest.octoprint.patch)

dirkvranckaert, 2023-06-01:
I simply downgraded, everything seems to be still working, only thing were the Octoprint camera settings that were gone (so need to re-enter the stream url)

crysxd, 2023-06-02:
> only thing were the Octoprint camera settings
Yes, when you moved from 1.8 -> 1.9 they were migrated into the bundled plugin and removed from the global settings. Those are not moved back when you downgrade

mikedmor, 2023-06-03:
Quick update. I have a semi working version ready for when Octoprint 1.9.1 releases.

To get the new update you first will have to update to version 0.3.1 to have access to the RC channel via "Software Update" in the settings. After getting version 0.3.1, you can then go to "Software Update" and switch MultiCam to the "Release Candidate" channel to get version 0.4.0rc1. 

Please note that there are still quite a few bugs that need to be worked out, it does however work better than 0.3.0 currently. 

Another note is that currently in Octoprint 1.9.0 this plugin forces you into safemode as there are some bugs with the core of octoprint that will require a patch. If you would rather not wait for 1.9.1 to release, you can manually update your core using the code posted to the Octoprint Repository: https://github.com/OctoPrint/OctoPrint/issues/4818

mikedmor, 2023-06-05:
rc4 was just released which addresses the issue regarding 1.9.0 and should now work for that version.

closed 2023-06-05 by mikedmor