mikedmor/OctoPrint_MultiCam #55: TODO: for Multicam 0.4.0 update

mikedmor, 2023-06-01:
Currently this plugin is under development to restore functionality for updates introduced in Octoprint 1.9.0. You can find the current progress under the 'rc' branch in this repository. Any assistance with this is welcomed.

- [x] Migrate python to utilize new webcam interfaces (requires some octoprint core fixes to load: will be released with 1.9.1)
- [x] Remove any code that adjusted the webcam dom previously
- [ ] Setup new multicam.jinja2 template for displaying the webcams
- [ ] Fix issue with restarting octoprint being required to show changes to the multicam settings
- [ ] Refactor plugin so that the settings and webcam displays are in two separate js files for easier maintainability.
- [ ] Add WebRTC, Hls, and Mjpg webcam types (see [classicwebcam](https://github.com/OctoPrint/OctoPrint/tree/master/src/octoprint/plugins/classicwebcam) implementation)

mikedmor, 2023-06-02:
Hey @chrisjq the easiest way is to just fork the [rc channel](https://github.com/mikedmor/OctoPrint_MultiCam/tree/rc) as that is where all the progress is being made. After making changes, you should be able to submit a PR from your forked repo and i can review those adjustments and pull them into the project. Please note the Octoprint BugFix when loading the plugin, as there is a bug with the core module in octoprint that is going to be fixed in version 1.9.1, but the updated core.js file can be found at that link and manually updated in the meantime. Thank you for offering assistance!

Once the multicam.jinja2 and a bit more of the multicam.js file have been updated i plan to release it as 0.4.0RC1 as that should be all that is left to get things working again at a bare minimum. The other todo in this list will be the full 0.4.0 release which should add some extra camera type features and refactoring things to simplify future improvements.

mikedmor, 2023-06-03:
Quick update. I have a semi working version ready for when Octoprint 1.9.1 releases.

To get the new update you first will have to update to version 0.3.1 to have access to the RC channel via "Software Update" in the settings. After getting version 0.3.1, you can then go to "Software Update" and switch MultiCam to the "Release Candidate" channel to get version 0.4.0rc1. 

Please note that there are still quite a few bugs that need to be worked out, it does however work better than 0.3.0 currently. 

Another note is that currently in Octoprint 1.9.0 this plugin forces you into safemode as there are some bugs with the core of octoprint that will require to be patched. If you would rather not wait for 1.9.1 to release, you can manually update your core using the code posted to the Octoprint Repository: https://github.com/OctoPrint/OctoPrint/issues/4818

mikedmor, 2023-06-05:
rc3 has now been released. This release addresses a bunch of issues, and implements HLC with a few bugs that need to be addressed. I unfortunately do not have the time to address RTC playback, but if there seems to be enough people looking for that I will look into adding it in a future update. 

There may be one more rc version that addresses a few additional bugs and QOL stuff, but besides that version 0.4.0 will be released shortly after Octoprint 1.9.1 is released which addresses a few bugs with octoprint needed to make this plugin work. 

Any additional feedback regarding HLS playback is welcomed.

opjose, 2023-06-05:
Does RC3 work with 1.9.0?

RC2 did not seem to.

Thanks!

mikedmor, 2023-06-06:
> Does RC3 work with 1.9.0?
> 
> RC2 did not seem to.
> 
> Thanks!

rc3 does not, however rc4 was released today which does address the issues with 1.9.0 and should be working now without the 1.9.1 patch.

cp2004, 2023-06-06:
Depending on OctoPrint 1.9.0 in setup.py will not block installation on an older version, but it has the side effect of also upgrading OctoPrint alongside this plugin. I would say that's not ideal, since the users won't know necessarily that they will also get the new OctoPrint version.

mikedmor, 2023-06-06:
> Depending on OctoPrint 1.9.0 in setup.py will not block installation on an older version, but it has the side effect of also upgrading OctoPrint alongside this plugin. I would say that's not ideal, since the users won't know necessarily that they will also get the new OctoPrint version. _Edit_: see #59

Thank you @cp2004 - I just pushed rc5 to prevent that from happening. I agree that is not good practice to automatically upgrade octoprint when my plugin is updated, I had assumed that was the correct way to achieve blocking installs by reviewing another plugin that did something similar. Appreciate the help in pointing that out.

closed 2023-06-09 by mikedmor