mikedmor/OctoPrint_MultiCam #61: MultiCam 0.4 constant streaming

szafran81, 2023-06-20:
Hi,

In my config MultiCam is streaming constantly (from the main OP page load, when I haven't even been on the control tab). This doesn't change after visiting Control tab, and then going to any other tab. I see constant streaming of the webcam image, which ends after closing OP tab, or opening another page in the same tab.

mikedmor, 2023-06-20:
This plugin is highly based upon the classic webcam implementation, which appears to do the same thing. Generally the webcam only unloads when octoprint runs a visibility update, however that does not appear to happen when changing tabs. This might have to be fixed in octoprint, however only 1 webcam feed will be active at a time, so it should not be problematic.

closed 2023-06-20 by mikedmor

szafran81, 2023-06-20:
> This plugin is highly based upon the classic webcam implementation, which appears to do the same thing. Generally the webcam only unloads when octoprint runs a visibility update, however that does [...]

I use OP since almost the time it got just a bit popular (I think ~10 years). And even then, when changing tabs the stream was ended after about 10-20 seconds after changing tabs, and it does this until this day (doesn't matter what cameras I've used... raspi cams, usb cams) - so you're completelly wrong in this regard. Since I use it frequently via my mobile connection this has a very bad affect on my data caps (basically more than usual data usage got me here).