mikedmor/OctoPrint_MultiCam #63: Cameras not loading in Octoprint native sidebar widget, some cams load OK on Edge, all cams load on FF. Go Figure!

marioPS, 2023-08-06:
Pretty much all in the subject field!
BTW: Classic webcam is disabled:
[image]

Config:
[image]

Both cams working in FF in 'Control' View:
[image]
[image]

But not in the sidebar:
[image]

And Edge doesn't like the webstreaming cam:
[image]
But FF is OK with it (Windows & Mac):
[image]

Config:
NozzleCam
http://[externalIP]:8081/camera/mjpeg
http://[externalIP]8081/camera/jpeg

JobbieCam
http://admin:camera@[internalIP]:80/video.cgi
http://admin:camera@internalIP]:80/image/jpeg.cgi

Let me know if you need any more info - & thanks in advance for your help!

Mikey!

LazeMSS, 2023-08-07:
This is a UI Customizer plugin error - im fixing it now.

marioPS, 2023-08-08:
Magnificent @LazeMSS - thank you for your fast response!

Edge now no longer "Loading webcam", but still does not stream JobbieCam.
[http://admin:camera@[internalIP]:80/video.cgi](http://admin:camera@%5BinternalIP%5D:80/video.cgi)
FF on both platforms working OK.

I hope this helps!

Mikey.

LazeMSS, 2023-08-15:
Should be fixed in laterst release of UI Customizer - please close if you agree @marioPS

marioPS, 2023-08-18:
Hi @LazeMSS Edge still having problem as previously explained.
Also, unable to get JobbieCam  http://admin:camera@[internalIP]:80/image/jpeg.cgi to provide still images for Timelapse …
Unknown if related.

LazeMSS, 2023-08-18:
What is Jobbiecam ?

marioPS, 2023-08-18:
Sorry! That's one of the two cams aimed at my printer: JobbieCam at bed (wifi camera), NozzleCam at the Nozzle (endoscope USB).
BTW: Just tested Chrome - same as Edge.
In both cases, I cannot switch cameras in the sidebar, but can in the main central view.

LazeMSS, 2023-08-18:
Strange I have no problems