mikedmor/OctoPrint_MultiCam #68: Pluginmanager offers 0.4.1 but installs 0.4.0

fly74, 2023-09-03:
Version seems to be wrong in zip:

```
2023-09-03 07:09:15,078   /home/op/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install https://github.com/mikedmor/OctoPrint_MultiCam/archive/0.4.1.zip --no-cache-dir
2023-09-03 07:09:16,352 > Collecting https://github.com/mikedmor/OctoPrint_MultiCam/archive/0.4.1.zip
2023-09-03 07:09:17,354 > Downloading https://github.com/mikedmor/OctoPrint_MultiCam/archive/0.4.1.zip
2023-09-03 07:09:17,355 > | 1.1 MB 5.0 MB/s 0:00:00
2023-09-03 07:09:17,355 > Preparing metadata (setup.py): started
2023-09-03 07:09:18,967 > Preparing metadata (setup.py): finished with status 'done'
2023-09-03 07:09:18,967 > Requirement already satisfied: OctoPrint in /home/op/OctoPrint/venv/lib/python3.11/site-packages (from OctoPrint-MultiCam==0.4.0) (1.9.2)
2023-09-03 07:09:18,967 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/op/OctoPrint/venv/lib/python3.11/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2021.2.23)
...
2023-09-03 07:09:23,689 > Successfully built OctoPrint-MultiCam
2023-09-03 07:09:23,689 > Installing collected packages: OctoPrint-MultiCam
2023-09-03 07:09:23,689 > Successfully installed OctoPrint-MultiCam-0.4.0
```

socmarin, 2023-09-03:
I have the same problem.

```
Actualizando, por favor espere.

+++++++++++++++++++++++++++++++++++
Actualizando ahora MultiCam a 0.4.1
+++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install https://github.com/mikedmor/OctoPrint_MultiCam/archive/0.4.1.zip --no-cache-dir
Collecting https://github.com/mikedmor/OctoPrint_MultiCam/archive/0.4.1.zip
Downloading https://github.com/mikedmor/OctoPrint_MultiCam/archive/0.4.1.zip (1.1 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 1.8 MB/s eta 0:00:00
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint-MultiCam==0.4.0) (1.9.2)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2021.2.23)
Requirement already satisfied: requests<3,>=2.28.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2.31.0)
Requirement already satisfied: colorlog<7,>=6.7.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (6.7.0)
Requirement already satisfied: zeroconf==0.39.4 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.39.4)
Requirement already satisfied: werkzeug<2.3,>=2.2.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2.2.3)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.8.0)
Requirement already satisfied: Flask-Login<0.7,>=0.6.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.6.2)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2.12.1)
Requirement already satisfied: PyYAML<7,>=6.0.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (6.0.1)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (21.3.0)
Requirement already satisfied: blinker<2,>=1.6.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (1.6.2)
Requirement already satisfied: flask<2.3,>=2.2.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2.2.5)
Requirement already satisfied: sarge==0.1.7.post1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.1.7.post1)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.2.6)
Requirement already satisfied: passlib<2,>=1.7.4 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (1.7.4)
Requirement already satisfied: sentry-sdk<2,>=1.19.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (1.24.0)
Requirement already satisfied: frozendict<3,>=2.3.7 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2.3.8)
Requirement already satisfied: markdown<4,>=3.4.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (3.4.3)
Requirement already satisfied: tornado<7,>=6.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (6.2)
Requirement already satisfied: Click<9,>=8.1.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (8.1.3)
Requirement already satisfied: future<1,>=0.18.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.18.3)
Requirement already satisfied: feedparser<7,>=6.0.10 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (6.0.10)
Requirement already satisfied: pyserial<4,>=3.5 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (3.5)
Requirement already satisfied: wheel in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.34.2)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2021.10.11)
Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.11.0)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (1.2.0)
Requirement already satisfied: semantic-version<3,>=2.10.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2.10.0)
Requirement already satisfied: OctoPrint-PiSupport>=2023.5.24 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2023.5.24)
Requirement already satisfied: regex in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (2022.10.31)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoPrint-MultiCam==0.4.0) (0.10.2)
[... cut ...]
Created wheel for OctoPrint-MultiCam: filename=OctoPrint_MultiCam-0.4.0-py3-none-any.whl size=18223 sha256=8c5f5c1fed5643e588fc0122496453d2a7de20003d3a94bd8b3285e4a3b1c50b
Stored in directory: /tmp/pip-ephem-wheel-cache-gtt6l9wd/wheels/9e/26/a8/910d4be58364f4adf3f7c2ea45f4d21425105e128990107a5d
Successfully built OctoPrint-MultiCam
Installing collected packages: OctoPrint-MultiCam
Successfully installed OctoPrint-MultiCam-0.4.0
La actualización ha finalizado correctamente y el servidor ahora se reiniciará.
```

victor7376, 2023-09-03:
Came here to see what the issue was as I'm having the same issue lol

mikedmor, 2023-09-03:
The update should have worked, but i did forget to increment the setup.py version number to indicate that. I just released 0.4.2 which should be available shortly and will address that. Apologies for the confusion.

closed 2023-09-03 by mikedmor

fly74, 2023-09-05:
Confirmed, works!