mikedmor/OctoPrint_MultiCam #69: Support for WebRTC?

thomasvnl, 2023-12-21:
Hi,

Just found this plugin. It seems really handy, but it does not want to work with the new libcamera stack which features webrtc on "host/webcam/webrtc", which is a shame as that is a far superior and slimmer streaming methodology compared to mjpeg. Could this plugin be modified to also allow for WebRTC streams to be loaded?

With kind regards,
Thomas

jacopotediosi, 2024-03-28:
As per #55, Multicam should already support WebRTC.

The implementation was copied from the [classicwebcam](https://github.com/OctoPrint/OctoPrint/tree/master/src/octoprint/plugins/classicwebcam) plugin, and allows you to use `webrtc://` or `webrtcs://` as the protocol in stream URLs.

Unfortunately, I imagine the current WebRTC implementation is broken, at least for those using the new camera stack, in both this and the classicwebcam plugins. By entering `webrtcs://192.168.1.26/nozzlecam/webrtc` as the stream url, I notice a ton of errors in the JavaScript console, while visiting `https://192.168.1.26/nozzlecam/webrtc` from my browser just works.

I suggest @mikedmor to look at the WebRTC implementation adopted by the [CameraStreamer-Control](https://github.com/cp2004/OctoPrint-CameraStreamer-Control) plugin, which instead works correctly, despite not supporting multiple cameras (yet?).

h3llb3nt, 2025-01-07:
so this is still broken today? im unable to get multicam to use the webrtc streams that work fine from my browser for my two cameras and have to use the old stream method still.

Gifford47, 2025-07-23:
Is it already fixed? Currently I cannot get it working with webrtc, no stream is loaded

closed 2026-09-11 by mikedmor