mikedmor/OctoPrint_MultiCam #71: take_webcam_snapshot() should accept a string as parameter

bchanudet, 2024-05-09:
Hello there,

I'm the author of the [OctoRant plugin](https://github.com/bchanudet/OctoPrint-Octorant), a plugin that sends messages from OctoPrint to Discord using Webhooks.

A user has reported a [bug](https://github.com/bchanudet/OctoPrint-Octorant/issues/79) where OctoRant is not sending any message when used in combination with MultiCam.

After investigation the crash happens when my plugin calls `take_webcam_snapshot()`. According to the [docs](https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.WebcamProviderPlugin), I call `take_webcam_snapshot()` with a string containing the webcam's name, but your plugin seems to expect a complete `Webcam` object, according to [`__init__.py:159`](https://github.com/mikedmor/OctoPrint_MultiCam/blob/277501232100dd033020f25e6229e1f250b6cdcf/octoprint_multicam/__init__.py#L159).

Looking further it seems that even OctoPrint's own timelapse plugin is [sending](https://github.com/OctoPrint/OctoPrint/blob/fed2dd7660c1587e2c280a30ed9d84ceffa1b89d/src/octoprint/timelapse.py#L771) a full `Webcam` object, yet at some other places it [sends only the name](https://github.com/OctoPrint/OctoPrint/blob/fed2dd7660c1587e2c280a30ed9d84ceffa1b89d/src/octoprint/server/util/tornado.py#L1526). Maybe I'll raise an issue on their side too.

bchanudet, 2024-05-09:
Given the doubts raised by reading OctoPrint source code about the usage of `take_webcam_snapshot()`, I've asked about it on the Discord server, I'll report back here when I have an answer.

NODeeJay, 2025-05-15:
This is connected (https://github.com/mikedmor/OctoPrint_MultiCam/issues/77). With OctoPrint 1.11 the wrongly implemented object has been fixed which renders the MultiCam plugin useless for snapshots or anything else that tries to access a snapshot. The author needs to fix it anyway to become compatible with OP >1.11 again.

jneilliii, 2025-05-17:
did you have multicam plugin installed prior to Feb 4th 2025? If so, the changes were merged then that fixes the string issue, but @mikedmor never published a release with the changes in version 0.4.3. You can force the install by copying/pasting this URL in plugin manager > get more > ...from URL to force the upgrade.
```
https://github.com/mikedmor/OctoPrint_MultiCam/archive/refs/heads/master.zip
```