mikedmor/OctoPrint_MultiCam #72: Not working with webrtc/mp4 streams generated by go2rtc

puterboy, 2024-06-10:
I am using go2rtc to generate streams for both a PiCamera v3 and a USB cam running on my RPi5.
Go2rtc seems to be a very powerful yet simple to implement streamer.

I can get it to work with Octoprint/MultiCam for mjpeg streams and legacy HLS streams.
But I can't get it to work for webrtc mp4 streams.

The webrtc url I have is of form:
```
http://mypi/api/stream.mp4?src=picam_h264
```
The stream plays just fine in my browser or in apps like mplayer and vlc.
But it fails to display in Octoprint.

(I would prefer not to use mjpeg since it's inefficient and and not HLS since it seems flakey)

So is there anything I need to do to get a  seemingly generic webrtc stream like the above to work with MultiCam?
What more generally is the status and limitations of webrtc for Octoprint?

Thanks

csm10495, 2024-12-28:
Same issue except for me I have an RTSP stream sent into go2rtc with HLS coming out. 

I'd like either to be able to have the RTSP stream directly rendered or HLS/WebRTC from go2rtc.

jacopotediosi, 2025-12-31:
I think that this issue could be a duplicate of #69.
Please see if [my fix](https://github.com/mikedmor/OctoPrint_MultiCam/issues/69#issuecomment-3701553664) works for you and attach here your JavaScript logs otherwise.