mikedmor/OctoPrint_MultiCam #74: OctoPrint 1.11.0 Changes WebcamCompatibility

jneilliii, 2025-02-03:
See here for related changes: <https://gist.github.com/foosel/4c57df27db637eb6136d70ab9ee6fb83#-webcamproviderplugintake_webcam_snapshot-has-gotten-its-parameters-fixed>

I believe this is going to impact the loading of this plugin.

mikedmor, 2025-02-03:
Thank you jneilliii for the heads up.

This adjustment appears to only affect the take_webcam_snapshot definition if I'm reading that post correctly. Changing `__init__.py` to the following around line 128-157 should fit it since this plugin is based on the classicwebcam for the most part, however I'm not able to debug or test this change at the moment to confirm as I'm in the middle of a move. If anyone wants to do a test and submit a PR ill get this updated, otherwise as soon as i get my setup running again ill debug this further and get something working.

```python
def take_webcam_snapshot(self, _):
        snapshot_url = self._get_snapshot_url()
        if not self._can_snapshot():
            raise WebcamNotAbleToTakeSnapshotException(self._webcam_name)

        with self._capture_mutex:
            self._logger.debug(f"Capturing image from {snapshot_url}")

            auth = self._settings.get(["snapshotAuth"])

            params = {
                "stream": True,
                "timeout": self._settings.get_int(["snapshotTimeout"]),
                "verify": self._settings.get_boolean(["snapshotSslValidation"]),
            }
            if auth in ["basic", "digest"]:
                username = self._settings.get(["snapshotUsername"])
                password = self._settings.get(["snapshotPassword"])

                if auth == "basic":
                    params["auth"] = requests.auth.HTTPBasicAuth(username, password)
                elif auth == "digest":
                    params["auth"] = requests.auth.HTTPDigestAuth(username, password)
            elif auth == "bearer":
                token = self._settings.get(["snapshotBearerToken"])
                params["headers"] = {"Authorization": f"Bearer {token}"}

            r = requests.get(snapshot_url, **params)
            r.raise_for_status()
            return r.iter_content(chunk_size=1024)
```

jneilliii, 2025-02-03:
You may need to change this line...

https://github.com/mikedmor/OctoPrint_MultiCam/blob/277501232100dd033020f25e6229e1f250b6cdcf/octoprint_multicam/__init__.py#L123

`snapshot = profile.get("snapshot", None) or ""`

I used this plugin as basis for my soon to be released go2rtc plugin (neat tricks with loading multiple templates, have recommendations for you), and after making the above change I was able to load my plugin without errors. Initially it didn't like None set on those profiles.

jneilliii, 2025-02-03:
I think your changes won't work. Ends up giving this error...

`AttributeError: 'MultiCamPlugin' object has no attribute '_get_snapshot_url'`

I made the above change I mentioned and was able to get the plugin to load into the RC, then added your changes mentioned above. What I understand the change is doing is making `provided_webcam` a string, rather than a full object, which means you'll have to do a lookup of the string provided to get a handle on your webcam.config settings. If I get a chance tonight I'll take a deeper look and provide a PR if I figure out anything. Probably just a simple type check on the passed provided_webcam argument.

jneilliii, 2025-02-04:
It seems that hacked together approach in the PR above should fix it. I ran it in both 1.10.3 and 1.11.0 and it successfully loaded and was able to take snapshots for timelapse generation.

PR #75 (2025-02-04): OctoPrint 1.11.0 Fixes
fixes issues related to the upcoming release of OctoPrint 1.11.0, resolves #74 I think.
closed 2025-02-04 by mikedmor